syntacore/scr1 #4: Assertions behavior, while reset, is incorrect

golikovav, 2017-05-29:
After every CPU reset (after end one test in series and start another) in ncsim console I see:

ncsim: *E,ASRTST (../src/pipeline/scr1_pipe_csr.sv,896): (time 132525 NS) Assertion scr1_top_tb.i_top.i_core_top.i_pipe_top.i_pipe_csr.SCR1_SVA_CSR_CYCLE_INC has failed
CSR Error: CYCLE increment wrong behavior
ncsim: *E,ASRTST (../src/pipeline/scr1_pipe_csr.sv,860): (time 132525 NS) Assertion scr1_top_tb.i_top.i_core_top.i_pipe_top.i_pipe_csr.SCR1_SVA_CSR_TIME_INC has failed
CSR Error: TIME increment wrong behavior

dp-sc, 2017-05-30:
We don't have ncsim and couldn't reproduce this type errors. These errors look like a racing. Could you try to change "clock properties" for the asserts from "posedge" to "negedge"?

golikovav, 2017-05-30:
Yes, It's works.

dp-sc, 2017-05-30:
Thank you, we will update all asserts to "negedge"

closed 2017-06-01 by dp-sc