syntacore/scr1 #8: Debug with Segger J-link

rockman182, 2019-01-29:
As described on page 13 of below document. J-link can work with SCR1.
[https://content.riscv.org/wp-content/uploads/2018/07/Shanghai-0925_AM_Syntacore_SCRx_SHA_day.pdf](url)
But my test failed , can't halt the cpu, connection failed.

The error message in  Ozone 2.61c
"JTAG chain detection found 1 devices:
#0 Id:0xDEB01001, IRLen: 04, Unknown device
CPU-TAP not found in JTAG chain"

dp-sc, 2019-01-29:
Could you please clarify, which SW stack you are using with J-link? 

It is currently supported with OpenOCD stack. 

Basic usage information is here:
https://github.com/syntacore/openocd/wiki/OpenOCD-for-sc_riscv32

Please, also read this: https://wiki.segger.com/OpenOCD 

Please, let us know if problem persists (additional information about environment may be useful). You can also write to scr1@syntacore.com

rockman182, 2019-01-30:
Based on the information you provide. SCR1 supports with OpenOCD stack only.
Please skip Ozone 2.61c tool issue, because it uses JLinkGDBServer.

However I try to use JLink + OpenOCD. It is still not working properly.
I can not halt CPU.
 
Below are detail log

ps: Using olimex ARM-USB-TINY-H + OpenOCD is okay.

sudo ./openocd -s ../tcl -f ~/openocd/tcl/interface/jlink.cfg -f ~/sc-riscv64-unknown-elf-gcc-20180126-linux64/openocd/tcl/target/syntacore_riscv32_v2.cfg

=======================================================
Open On-Chip Debugger 0.10.0+dev-01006-gbc11293 (2017-12-04-16:11)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
init_targets
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
adapter speed: 2000 kHz
trst_and_srst separate srst_gates_jtag trst_push_pull srst_open_drain connect_deassert_srst
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
scr.cpu
Info : No device selected, using first device.
Info : J-Link V9 compiled Oct 25 2018 11:46:07
Info : Hardware version: 9.10
Info : VTarget = 3.275 V
Info : clock speed 2000 kHz
Info : JTAG tap: syntacore_riscv32_v2.cpu tap/device found: 0xdeb01001 (mfg: 0x000 (<invalid>), part: 0xeb01, ver: 0xd)
Info : IDCODE=0xDEB01001 DBG_ID=0x00820000 BLD_ID=0x18110700
Info : Listening on port 3333 for gdb connections

===================================================================

telnet localhost 4444

Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* scr.cpu            syntacore_riscv32_v2 little syntacore_riscv32_v2.cpu running
> targets scr.cpu
> halt
Target not halted
in procedure 'target'
> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* scr.cpu            syntacore_riscv32_v2 little syntacore_riscv32_v2.cpu running
> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* scr.cpu            syntacore_riscv32_v2 little syntacore_riscv32_v2.cpu running
> halt
Target not halted
in procedure 'target'
DBG_STATUS == 0xE00300DC
Lock detected: 0xE00300DC
========= Try to unlock ==============
DBG_STATUS == 0xA00300DC
Lock with lock_context=0x00000034 repaired: 0xA00300DC
Hart errors detected: 0xA00300DC
Core status: 0x800000C0
>

dp-sc, 2019-01-30:
Hey rockman, - some additional info about your env is still needed, and could you please send it to the scr1@syntacore.com? It's likely a SW configuration-related support question, which probably doesn't belong to the scr1 issues section.

Anyway, here is further info about your env which will help to root-cause your issue:

1) Could you clarify what is the HW platform being used? Is it one of SCR1 SDKs (FPGA board + FPGA image from our GitHub repo), or your custom hardware?

2) Which SW is running when you're trying to connect and halt SCR1? Is it our SDK bootloader or your own executable code?

3) If this is your custom design - what are the SCR1 core clock and JTAG frequencies? Is the requirement {fCoreClkFreq/fTck > 12} met?

4) Could you please provide more verbose log with -d3 option (sudo ./openocd
-d3 -s ../tcl -f ~/openocd/tcl/interface/jlink.cfg ....)? again, mail is preferable.

rockman182, 2019-02-14:
I borrow a JLINK with hardware version 10.1. 
Everything is okay.

closed 2019-02-14 by rockman182