syntacore/scr1 #9: Coremark test.

jasonlee1001, 2019-03-14:
Hi,
  I did all the steps and type:
  make run_verilator BUS=AXI ARCH=IM IPIC=0
  ISA is pass, Dhrystone is pass.
  But the Coremark test seems kind of odd ! The partial message as follow: (in Line 363)
  -----------
358 CoreMark 1.0
359 2K performance run parameters for coremark.
360 CoreMark Size    : 666
361 Total ticks      : 3533
362 Total time (secs): 0
363 ERROR! Must execute for at least 10 secs for a valid result!
364 Iterations       : 1
365 Compiler version : GCC7.1.1 20170509
366 Compiler flags   : -O2 -funroll-loops -fpeel-loops -fgcse-sm -fgcse-las
367 Memory location  : STATIC
368 seedcrc          : 0xe9f5
369 [0]crclist       : 0xe714
370 [0]crcmatrix     : 0x1fd7
371 [0]crcstate      : 0x8e3a
372 [0]crcfinal      : 0xe714
373 Errors detected
374 ^[[0;32mTest passed^[[0m

Can someone help to solve this ? Many thanks.

Jason Lee

dp-sc, 2019-03-15:
Dear Jason Lee, 
You need to increase ITERATIONS parameter while you run coremark(by default ITERATIONS=1)
Just type:
`make run_verilator BUS=AXI ARCH=IM IPIC=0 ITERATIONS=31000`
But this will significantly increase simulation time. 
Dmitri

jasonlee1001, 2019-03-15:
Dear Dmitri,
  Thank you. It starts to run and really takes long long time.
  Sorry, again a question. Is this ITERATIONS belong the parameter for
Verilator, or in *.sv , *.c , or *.h (I grep all these three kinds of file
and not seeing this word) ?

Jason Lee

dp-sc, 2019-03-15:
Dear Jason Lee,
Parameter ITERATIONS belong to coremark (more detailed in https://github.com/eembc/coremark/blob/master/README.md).
In our case, if you set the value of ITERATIONS = 31000, this means run the benchmark for 31000 iterations and
 will correspond to approximately 10 seconds
Dmitri

jasonlee1001, 2019-03-15:
Dear Dmitri,
  Thank you for replying my question so quickly and forgive my poor English which is not my native language...
  I had heard Mr. Pavel Khabarov speak in RISC-V workshop in Taiwan. I just want you guys to know you did a good job and we are evaluating the SCR1. Other series(SCR3/4/5) also be candidates.

closed 2019-03-16 by jasonlee1001