syntacore/scr1 #12: Which files are need to be synthesized?

jasonlo0509, 2019-08-15:
Dear Sir,

I am trying to synthesize scr1 using dc_shell. 
As all files are written in system verilog, I am wondering which files need to be synthesized?

Currently I put all files list in src/axi_top.files and src/core.files to analyze command in dc_shell

Thank you,
Yun-Chen Lo

ar-sc, 2019-08-17:
hi Yun-Chen, yes, these are correct lists.

If you are doing synthesis, please pay attention to library-specific cells, which need to be set properly i n the instance. Depending on the configuration you use, you may need to modify these:
scr1\src\core\primitives\scr1_cg.sv - for library specific clock gating element
scr1\src\top\scr1_dp_memory.sv - for library specific memory modules
Will add section on that in the user guide in the coming release.

cheers, a

eb-sc, 2019-10-24:
We've closed this issue. Please open a new one if you have any further questions.

closed 2019-10-24 by eb-sc