syntacore/scr1 #13: Are there any big modification for correct synthesis results?

jasonlo0509, 2019-08-15:
Hello,

I am trying to synthesize and tap-out a prototype chip using scr1. Many thanks to the nice work and kind continuous update!

Since I am cloning scr1 half year ago, I am wondering if there's any important updates for correct synthsized results. Because I stuck at unknown standard delay format annotation failure after generating .sdf files using dc_shell and run gate-level simulation using irun.

Many Thanks!

Best Regards,
Yun-Chen Lo

dp-sc, 2019-08-17:
Hello Yun-Chen, you are welcome and thank you for using scr1.

Could you please list the specific SHA you are currently based on? We try to keep informative descriptions for commits, and don't recall any critical updates recently, - but to confirm, we'd need a specific revision. Although, as a rule, we always recommend to use the most recent version.

To comment on the synthesis errors you observe, we'd need additional details. Please, send your logs to the scr1@syntacore.com. With that, various companies took scr1 through TO mutiple times, including older revs since 2017, - using different flows and w/o reporting any major issues.

Dmitri

jasonlo0509, 2019-08-17:
Dear Dmitri,

I use 2 versions to synthesis: 30b6ad1777333820de80a440fbc9ed96298cf19c and a previous version that hold workshop in NTHU, Hsinchu, Taiwan.

My main problem encountered is that because most of the DFF in the design use async reset, e.g. (negedge rst_n, posedge clk). When I use dc_shell to synthesized under TSMC process. I encounter gate-level simulation error called recrem<removal> and lead to ASSERTION fail "scr1_top_tb_axi.i_memory_tb.rw_if[1].SVA_TBMEM_X_ARVALID has failed"

Are there any command that I need to pay attention to when using dc_shell to synthesize the design?

Sure, I will send detail information to the email.

Thank you again!
Yun-Chen Lo, NTHU

eb-sc, 2019-10-24:
We've closed this issue. Please open a new one if you have any further questions.

closed 2019-10-24 by eb-sc