syntacore/scr1 #14: Non atomicity of traps on SCR1 core ?

elie1968, 2019-09-20:
Hello SCR1 team,

I have doubts about trap atomicity on SCR1 core, that is:
I am able to raise exceptions on incorrect bus transactions even though MSTATUS = 0x00001800 and MIE = 0x0
Also, after reset MSTATUS=0x00001880, that is mstatus.mpie = '1' (machine previous interrupt enable) 
Why is bit 7 '1'? should it be '0' after reset.

As far as I understand the The RISC-V Instruction Set Manual / Volume II: Privileged Architecture (but I may be wrong) traps are allowed in machine mode only when mstatus.mie = '1'

Therefore a trap in a trap can be raised leading to an infinite loop, where having disabled trap (in the trap handler for example) should enforce atomicity of traps.

I have attached a VCD file where I can raise a trp on bus error when traps have not been enabled, and even trap in a trap infinite loop.

I may be wrong in my analysis, in this case, how can I enforce atomicity of trap in trap handlers ?
If it is effectively a hardware bug, is there a quick patch that I can very quickly apply before synthesis ?

Thank you very much,
Kind regards

Jean-Marc
[scr1-iaf.vcd.gz](https://github.com/syntacore/scr1/files/3635048/scr1-iaf.vcd.gz)
[trace_csr_0.log](https://github.com/syntacore/scr1/files/3635056/trace_csr_0.log)

dp-sc, 2019-09-23:
Dear Jean-Marc, 

We’ve reviewed the case and think the observed behavior with the infinite loop can rather be attributed to the some SW bug. Please, find justification below.

First of all, let’s look at the "Privileged Architecture specification".  Section <3.1.6.1 Privilege and Global Interrupt-Enable Stack inmstatusregister> says:
“_Global interrupt-enable bits, MIE, SIE, and UIE, are provided for each privilege mode. These bits are primarily used to guarantee atomicity with respect to interrupt handlers in the current privilege mode._”

The incorrect bus transaction rises synchronous exception, not an interrupt. We don’t think one can (and should) try preventing synchronous exceptions like that.

The infinite loop you are observing is most likely a SW, not a HW bug. 
Let’s consider the following example: The MTVEC points to a non-existing memory, and the memory subsystem causes “Instruction Access fault” exception after fetch attempt from this address.
Then, for this case, any exception or interrupt will lead to another fetch attempt from the MTVEC address. And this second access will cause another “Instruction Access fault” and so on.
We don’t think one could ignore this exception, but the case itself is valid - because you can’t fetch a valid instruction from a non-existing memory. Hence, this infinite fetching from the MTVEC address is a valid behavior.

About the MSTATUS, the spec says in section <3.3 Reset>:
“_Upon reset, a hart’s privilege mode is set to M. The mstatus fields MIE and MPRV are reset to 0. … All other hart state is unspecified_”
So, there is no contradiction here either.

Dmitri

elie1968, 2019-09-24:
Hello Dimitri,

I admit your argument, however on many other processors (like sparc-V8, which I have used for a long time), exceptions (synchronous or asynchronous) are disabled by the hardware when one is raised and you have to re-enable it (at your own risk !) in the trap handler.
On ARM processor, which I also have used (A53/R4/M4/M0) a lot, as far as I remember it is the same.
Also on all these processors, at hardware reset, exceptions and interruptions are all disabled.

I'll do a quick software fix, even if it won't solve the root of the problem.

I close the issue.

Thank you again for your support
Kind regards

Jean-Marc

closed 2019-09-24 by elie1968