syntacore/scr1 #16: PULPissimo integration

MarekPikula, 2019-10-17:
I'm evaluating different RISC-V cores and I started with ones available for PULPissimo (ie. RI5CY and Ibex). Afterwards I've found out your project and thought that to make a fair performance comparison I could just use scr1 in PULPissimo.

I've written a wrapper to use with PULPissimo. Would you like me to push my changes to this repo and make scr1 an option in official PULPissimo repo?

eb-sc, 2019-10-24:
Potentially, we don’t mind, - please, submit your pull request for review. We may need to keep PULPissimo-related changes in a dedicated branch, - hope that’s fine?

Best regards,
Kate

MarekPikula, 2019-11-07:
I'll let you know here, when there will be some development in upstream PULPissimo.

eb-sc, 2019-11-07:
Ok, thank you.
If you still plan to include the file for PULPissimo builder or any other, you could create another pull request. The branch name is "pulpissimo_integration".

closed 2019-11-07 by eb-sc

MarekPikula, 2019-11-07:
Sure, thank you for merging.