syntacore/scr1 #17: Compliance_breakpoint test failed when I disable breakpoint module

r901042004, 2019-10-17:
In [breakpoint.S](https://github.com/riscv/riscv-compliance/blob/2fdea4a96cac9846d426242079be9528c267d877/riscv-test-suite/rv32mi/rv64mi/breakpoint.S#L23) of riscv-compliance, `csrw tselect, x0` will cause illegal instruction exception when the breakpoint module is disabled.
I think the exception is correct because the `tselect` CSR isn't implemented in such configuration.
Skipping the breakpoint test file in [scr1_top_tb_ahb.sv](https://github.com/syntacore/scr1/blob/master/src/tb/scr1_top_tb_ahb.sv#L333) (axi is similiar) is an easy method.

```systemverilog
`ifdef SCR1_BRKM_EN
    if (is_compliance(test_file)) begin
        if (get_ref_filename(test_file) != "breakpoint") begin
`endif // SCR1_BRKM_EN
            i_memory_tb.test_file = test_file;
            i_memory_tb.test_file_init = 1'b1;
            $write("\033[0;34m---Test: %s\033[0m\n", test_file);
            test_running <= 1'b1;
            rst_init <= 1'b1;
`ifdef SCR1_BRKM_EN
        end
    end
`endif // SCR1_BRKM_EN
```
I didn't add string comparison of Verilator's version in above code.

eb-sc, 2019-10-25:
If we read this correctly, you’ve made an SCR1 configuration without SCR1_BRKM_EN and want to exclude the riscv-compliance "breakpoint" test from the list for simulation using Verilator.

Unfortunately, the Verilator does not support all simulation features of the SystemVerilog language (hope this will be changed soon), in particular, the comparison of two strings. You could read more about supported features in the "Verilator manual": https://www.veripool.org/projects/verilator/wiki/Documentation

As a solution, we suggest you to manually exclude this test from the list. To do this, please edit the file ./sim/tests/riscv_compliance/Makefile by adding a new line to it
`cut_list      += breakpoint`
As you could see here the same procedure was done for several incompatible riscv-compliance tests https://github.com/syntacore/scr1/blob/9bc6cbbbd3670040720d8e39e7f569b3f2796df6/sim/tests/riscv_compliance/Makefile#L71
 

Best regards,
Kate

r901042004, 2019-10-26:
Actually, I used ncsim, vcs & modelsim for simulation. The method I proposed works in these simulators except Verilator apparently. Put the test in the cut list is the easiest way. However, the situation may need to be addressed in README or document to remind users.

eb-sc, 2019-11-07:
Thank you for your comment. We will definitely include such a description in the next README and UM release.

closed 2019-11-07 by eb-sc