syntacore/scr1 #28: riscv_isa march option is fixed even when changing ARCH

leduchuybk, 2021-07-05:
At line 4 in the makefile of riscv_isa test
I can see that it is fixed with IMF as march for toolchain.
Will it affect the test result when i setup test for CFG=BASE (RV32IC)?

dp-sc, 2021-07-14:
Hi, Leduchuybk
This specific fixed march does not affect which isa tests will be selected for a given configuration, nor the correctness of their assembly 
So, we expect correct test results on any supported configuration (but without RVE extension). 
Syntacore team

leduchuybk, 2021-07-16:
After taking a closer look at environment, i can see that individual test of each instruction is added at 
scr1/sim/tests/riscv_isa/rv32_tests.inc
Thank you for your respond

closed 2021-07-16 by leduchuybk