syntacore/scr1 #29: Differences in executing instruction between configuration

leduchuybk, 2021-07-16:
At 6.2 in scr1_eas.pdf, you guys explained the main difference between configuration MAX, BASE, MINN is the insertion of instruction queue. Will this queue affect the clock cycle needed to execute one instruction between configurations?

dp-sc, 2021-07-16:
Hi Leduchuybk,

The queues do not affect the execution time of any instruction in ALU.
More details about execution time for the instructions you can find in corresponding tables in “6.2. Pipeline configurations”.

Syntacore Team

leduchuybk, 2021-07-18:
From what I see from waveform,  execution time for [Add] instruction in different configurations is the same.
But the specification defines latency and throughput of one instruction is different between configurations.
What is the difference between configurations besides existence of the queues?

dp-sc, 2021-07-21:
Hi Leduchuybk,

Latency depends on the queues presence. 

If you’ll refer to EAS, in Section 6.2. Pipeline configurations, it says: “Latency column shows the number of clock cycles from the start of the instruction fetching until the instruction retirement.” 

Syntacore Team

closed 2021-08-03 by dp-sc