syntacore/scr1 #31: cannot generate random number on scr1 with pre-build binary tool

odingo, 2021-07-23:
Hi,
I'm trying to run simple C code on scr1, but failed to generate random number.
I used 'rand()' in the C code and compiled it with the pre-build binary tool which was downloaded from http://syntacore.com/page/products/sw-tools.
But the 'rand()' always returned zero value.

odingo, 2021-07-26:
The workaround now is to use -Wl,--wrap=srand,--wrap=rand with compiler and to write my own srand(), and rand() functions.

ir-sc, 2021-08-03:
> Hi,
> I'm trying to run simple C code on scr1, but failed to generate random number.
> I used 'rand()' in the C code and compiled it with the pre-build binary tool which was downloaded from http://syntacore.com/page/products/sw-tools. (in Linux)
> But the 'rand()' always returned zero value.

Hello!
We have checked and have not found any problems with the rand function. Please describe what you are doing in more detail (we are interested in the compiler keywords, linker script and minimal code).

BR,
Irina

odingo, 2021-08-13:
# CPU architecture
ARCH := ec
ABI := ilp32e

# Tool chain
export CROSS_PREFIX  ?= riscv64-unknown-elf-
export GNU_CC      ?= $(CROSS_PREFIX)gcc
export GNU_OBJDUMP ?= $(CROSS_PREFIX)objdump
export GNU_OBJCOPY ?= $(CROSS_PREFIX)objcopy

LINKER_SCRIPT = $(CODE_DIR)/common/link.ld

FLAGS = -O0 -funroll-loops -fpeel-loops -fgcse-sm -fgcse-las -g -Wl,--wrap=srand,--wrap=rand
CFLAGS_COMMON = -static -std=gnu99 -fno-common -fno-builtin-printf
CFLAGS_ARCH = -Wa,-march=rv32$(ARCH) -march=rv32$(ARCH) -mabi=$(ABI)

GNU_CC_FLAGS = $(FLAGS) $(CFLAGS_COMMON) $(CFLAGS_ARCH) -D__ASSEMBLY__=1 -DRISCV_SCR1 -DSIM_SEED=$(SEED)
LDFLAGS ?= -nostartfiles -nostdlib -lc -lgcc -march=rv32$(ARCH) -mabi=$(ABI)

odingo, 2021-08-13:
[image]
Linker script is modified from the example code of scr1 database but not many changes.

dp-sc, 2022-04-22:
Hello [odingo](https://github.com/odingo)

We couldn’t reproduce the issue.  Could you sent us the information about your toolchain. To check the toolchain version, please run :

````
> riscv64-unknown-elf-gcc -v

````

Syntacore

closed 2022-05-12 by dp-sc