syntacore/scr1 #34: Per-instance customization via module parameters

alexforencich, 2022-02-02:
It looks like customization of the core configuration is all done via preprocessor macros instead of module instance parameters.  The problem with this is that it's not possible to include multiple instances of the core in a design with different settings - for instance, maybe a design needs a couple of smaller cores with several features disabled, in addition to a higher-performance core with a different set of features enabled.  It seems like most of the preprocessor macros can probably be replaced with module parameters, which would make the core significantly more flexible.

dp-sc, 2022-04-21:
Hi [alexforencich](https://github.com/alexforencich)

Thank you for your comment. 
Unfortunately, the configuration with parameters doesn’t work for all cases. For example you couldn’t configure the input/output ports (scr1_pipe_csr.sv) :
\`ifdef SCR1_IPIC_EN
    // CSR <-> IPIC interface
    output  logic                                       csr2ipic_r_req_o,           // IPIC read request
    output  logic                                       csr2ipic_w_req_o,           // IPIC write request
    output  logic [2:0]                                 csr2ipic_addr_o,            // IPIC address
    output  logic [\`SCR1_XLEN-1:0]                      csr2ipic_wdata_o,           // IPIC write data
    input   logic [\`SCR1_XLEN-1:0]                      ipic2csr_rdata_i,           // IPIC read data
\`endif // SCR1_IPIC_EN

If you need to use cores with different configuration you can use “rename” technics:
All files/modules/types/{struct elements} have prefix “scr1_” in the their name.
All parameters/macros/{enum elements} have prefix “SCR1_” in the their name.

To use several cores with different configuration in one design, you need to rename for each configuration:
 {scr1_/SCR1_} -> {prefix_/PREFIX}

Syntacore

alexforencich, 2022-04-21:
Parameters work fine for that case, trivially.  Remove the ifdefs, tie off the outputs, and disconnect the inputs based on parameter values with normal if statements and ternary operators.  Sure, the ports will still be there, but since they're disconnected they will have no affect on anything.

dp-sc, 2022-04-22:
Hello [alexforencich](https://github.com/alexforencich)

````
Sure, the ports will still be there, but since they're disconnected they will have no affect on anything
````

Not exactly. 
First of all, during synthesis/simulation you will get additional warnings like “input signal is not used” and “signal is tied to constant”. And the secondly, the verification coverage will be dropped down.

Another example – is the using ifdef/else/endif in the combinatorial blocks(scr1_pipe_csr.sv) or in the enumeration type declaration(scr1_riscv_isa_decoding.svh).

Thank you for your time, but we are not planning to change the RTL of the core in accordance with your proposal.

Syntacore

closed 2022-05-12 by dp-sc