syntacore/scr1 #37: how to modify tcm‘s address

snlkah, 2022-07-08:
We modified the tcm_mask and tcm_pattern of tcm to 32'hfffe0000 and 32'h00000000 respectively, so that the address of the tcm instruction is 32'h00000000 to 32'h00001fff. However, during the actual operation, the address exceeds 32'h00001500, and the scr1 displays the x state.We wonder if we want to modify the address of tcm ,We only need to modifythe  tcm_mask and tcm_pattern.

achuykov-sc, 2022-07-11:
Hi @snlkah,
Could you please check that you initialize your memory before read? By default, the memory is not initialized, and read before write may cause 'x' propagation. For example, to fix the issue, you would write something to that address during application start-up. 
And please double check that you modify your linker script (memory map) according your TCM changes.

Best regards,
Alexander

snlkah, 2022-07-12:
Hi @achuykov-sc 
Thank you for your reply.I have checked that i initialized my memory before read. And now the symptom is that I can read and write normally at first, but will enter the x state after a period of time.
And I want to confirm that when I use the tcm module, I assign the value 32'hfffe0000 to SCR1_PORT1_ADDR_MASK and the value 32'h00000000 to SCR1_PORT1_ADDR_PATTERN in the router module respectively , i think which means modify my linker script (memory map) according my TCM changes.Is it right?
And if i want to use the tcm module,do i only need to open SCR1_TCM_EN option?
Best regards,
snlkah

snlkah, 2022-07-19:
Hi @achuykov-sc ,
I'm sorry for replying you so late. For some reason we can't provide the wave file, but i found out that the reason is the SCR1 core jumped to the address beyond the address boundary when executing the ret instruction, which caused The core cannot operate normally, do you know how to solve this problem?
Best regards,
snlkah

achuykov, 2022-07-19:
Hi @snlkah,

Thanks for update. It seems, you have a corruption of the memory. It means, another part of application corrupts the stack which contains return address. For example, the IRQ handler suddenly writes out of the array boundary and corrupt the return address in the stack. 
 
My suggestions for you:
1. Please check your software and pay attention on memory accesses
Please be accurate with compiler warnings, they may have some recommendations with protentional issues in code 

2. Please check stack size and memory allocations 
You could use `-fstack-usage` option for gcc to analyze stack size requirements 

3. Could you please enable tracing in core? 
Thus, you could analyze the trace and find the root cause   

Best regards,
Alexander

closed 2022-07-25 by snlkah