syntacore/scr1 #38: Will there be any plan for adding the cache module in the core later?

snlkah, 2022-08-09:
Will there be any plan for adding the cache module in the core later?

achuykov-sc, 2022-08-09:
Hello [snlkah](https://github.com/snlkah),

We are not supporting caches in SCR1 cores. And we do not have plans to add them. 

Best regards,
Alexander

snlkah, 2022-08-10:
Hi @achuykov-sc 
Thank you for your reply.Do you have any difficulty in adding the cache function? Because I am exploring whether I can add the cache function to SCR1 by myself.
Best regards,
snlkah

achuykov-sc, 2022-08-15:
Hi @snlkah, 

It is hard to predict complexity because it depends on features that you want to have in cache. 
For simple systems, you could add the block between memory interface and core. But this idea is far from real life, where the cache is tightly coupled with core pipeline. And it uses internal buses to read/write data and fetch instructions. 
Of course for normal operations, you should add control and status registers (at least) to maintain the cache. Another challenging thing, is the cache itself, it's structure and logic. 

Best regards,
Alexander

closed 2022-08-24 by snlkah