syntacore/scr1 #41: Syntax error

m-spr, 2022-11-24:
Hello,

I did the "make" and face multiple errors related to the "Behavior check". 

[image]

 I tried to comment these part, but of course, the design failed in Testing as follow.

[image]

I have also searched for the problem there was a few solutions on the internet, but non of them work. 
Can you please help me to solve this?

Thank you in advance

achuykov-sc, 2022-11-24:
Hello @m-spr,

Could you please share your full log and configuration file? 

Best regards,
Alexander

m-spr, 2022-11-24:
For sure. Here you are.

Thank you.

[log.zip](https://github.com/syntacore/scr1/files/10086322/log.zip)

achuykov-sc, 2022-11-24:
Thanks! It seems, you have issues with "verilator". What is the version of "verilator" do you use? 

About config file, I was not clear. I mean "Architecture description file": includes/scr1_arch_description.svh. Did you modify it? 

And what are parameters for "make" command? 

BR,
Alexander

m-spr, 2022-11-24:
I have not changed anything in the Architecture description file. I tried to do "make" with the ModelSim option, and it seems that it is working. Although, I changed the "riscv64-unknown-elf-" to "riscv32-unknown-elf-", and now I have some problems with that. It reports "Test failed" because the "riscv64-unknown-elf-readelf" is not found. Where else should I fix this?
The log file is attached to this message.

[log.TXT](https://github.com/syntacore/scr1/files/10086483/log.TXT)

Thank you for your help

closed 2022-11-25 by m-spr