syntacore/scr1 #42: Syntacore custom modification problems

GARIIKK, 2022-12-08:
Hello, 
I have taken bare SCR1 core, then add my own RAM and ROM and create wishbone wrapper in order to use wishbone as a communication BUS. I have also generated .memh files with my soft for every extension I needed and load it to ROM. Also instantiate some necessary defines in my tb.  But I have some problem with instruction receiving from ROM. PC stucks in 0x1C0 value immediately after first instruction request. I attached some screenshots of my defines, .memh file and waveforms with stucked  PC
[image]
As you can see on screenshot below, boot addr started from 0x80, so I changed SCR1_ARCH_RST_VECTOR = 0x80.
[image]
To summarize, I can read first instruction from ROM but PC stucks even before cpu2mem request.
[image]
Could you please advise me something, probably I forgot to instantiate some defines or so?

achuykov-sc, 2022-12-08:
@GARIIKK, 

Could you please check the linker script for your application (CoreMark)? 
The default entry point is 0x200. So, if you had changed the reset vector address, you should change linker script according to your changes. 

Best regards,
Alexander

GARIIKK, 2022-12-12:
Hello @achuykov-sc ,
Sorry for delayed answer. I have found the problem. It is closely connected with my SyntaCore2Wishbone bridge. I reread your documentation on AHB and AXI bus. But I still cant understand how I should do this communication correctly. I did some changes with delaying(pipelining) of some request/response signals to/from Wishbone BUS. After this modification Core starts execute the instruction, writes and reads from Data memory. But still does it incorrectly(compare to another RISC-V core which 100% works correctly with the same soft, linker script, etc.). Could you please describe this SyntaCore2BUS communication in more details?
Regards,
Ilya

closed 2023-01-12 by achuykov-sc