syntacore/scr1 #44: SCR bootloader’s banner does not show up after I RESET the Arty A7-100T board.

shipherd, 2023-01-12:
Hi,

I am trying to use the SCR1 SDK on **_Arty A7-100T_** board and was following the instructions given by **_arty_scr1_guide_en.pdf_**.
I can flash the memory with the part **_S25FL127S_** (refer to 3. FPGA Configuration Flash Programming). The DONE LED is on when flashing the ROM, and it is still on when the flash progress is done.
After I press the PROG button, the DONE LED becomes off and **never** come back to on state unless I reflash the ROM. I also did not receive the banner message as described in section 4 when the flash is done, no matter I press the PROG nor RESET buttons after flashing the ROM. (nothing was received in the serial terminal)

[image]

What could be wrong? Thank you. :-)

achuykov-sc, 2023-01-12:
Hi @shipherd,

Thanks for details but I have extra comments and questions for you? 
1. Please check that prebuild FPGA image is working for you: [scr1-sdk/images/arty/scr1 at master · syntacore/scr1-sdk (github.com)](https://github.com/syntacore/scr1-sdk/tree/master/images/arty/scr1)
2. If you build bitstream and MCS by yourself, could you please check that bitstream is working and after that switch to MCS
3. Please make sure that you select SPI FLASH chip correctly. Unfortunately, different chips may be on board (S25FL127S or S25FL128S): https://files.digilent.com/resources/programmable-logic/documents/S25FL127S_PCN.pdf
4. And I'm pretty sure, our document was written for A35t which is slightly different with A100t

Best regards,
Alexander

shipherd, 2023-01-12:
Hi Alexander,

1. I was using the prebuild images. (which was cloned directly from the repository)
2. Not using the self-build images.
3. The SPI FLASH is labeled as shown below:
[image]
4. Have you tested them on A7-35t before ? (Not the old Arty 35t)

Thank you :-)

achuykov-sc, 2023-01-12:
Hi @shipherd,

The scr1-arty-fpga.7z (https://github.com/syntacore/scr1-sdk/blob/master/images/arty/scr1/scr1-arty-fpga.7z) contains bitstream (aka arty_scr1_new.bit) in addition to MCS. Could you please try the bitstream (arty_scr1_new.bit) first without MCS programming? 

Please note, the A100t is incompatible with A35t bitstream. 

Best regards,
Alexander

shipherd, 2023-01-12:
Hi @achuykov-sc ,

So, I am supposed to recompile everything for 100t board?

Thank you. :-)

shipherd, 2023-01-12:
Hi @achuykov-sc,

I had recompiled the Block Design project and the banner came up.

Thank you for your help :-)

closed 2023-01-12 by shipherd

I3ernhard, 2024-08-24:
Hi @shipherd, how did you managed it to run it on the Arty A7 100T Board?

Thanks!