syntacore/scr1 #45: The behavior of Stepping through DCSR.step and DMCONTROL.resumereq

shipherd, 2023-02-03:
Hi,

I was wondering why do I have to Halt the hart again after a step and before the next step? I have tested that the hart must be halt (DMCONTROL.haltreq) again after a resume request (DMCONTROL.resumereq), otherwise the next step (resume request) would not work.

To my understanding on _"riscv-debug-release.pdf, section 'Appendix B.5'"_, should not the hart be halt automatically after the resume request?

[image]

Thank you. :-)

achuykov-sc, 2023-02-22:
Hi @shipherd,

Please apologize in reply delay. I had forwarded your question to R&D. Let wait for response from them. 

Best regards,
Alexander

dk-sc, 2023-05-16:
Hi @shipherd ,
Sorry for a delayed reply. We got answer from our R&D:

_Actually, after resuming a debugger must wait until DMSTATUS.resumeack = 1, and then clear it. Otherwise, if the HART is halted after single step, and DMSTATUS.resumeack = 1 so far, writing of 1 to DMCONTROL.resumereq again virtually does nothing (as DMCONTROL.resumereq already =1 at that moment from the previous resuming).

To clear DMSTATUS.resumeack, a debugger must write DMCONTROL.resumereq = 0. After that next resuming is possible.

In your case, writing of 0x80000001 to DMCONTROL just after resume not only sets haltreq bit (which is useless in that context), but also clears resumereq (and DMSTATUS.resumeack), and that fulfills the requirement above. So, after resuming you should write DMCONTROL = 0x00000001 instead of 0x80000001._

Best regards,
Daria

closed 2023-05-19 by shipherd