syntacore/scr1 #50: Error: unrecognized opcode `fence.i'

cartersmith15, 2023-07-31:
I am trying to get the simulator working but when I run the make command I am met with the following issue:

/home/csmith/carter_scr1/sim/tests/riscv_isa/../../../dependencies/riscv-tests/isa/rv32ui/../rv64ui/fence_i.S:24: Error: unrecognized opcode `fence.i'
/home/csmith/carter_scr1/sim/tests/riscv_isa/../../../dependencies/riscv-tests/isa/rv32ui/../rv64ui/fence_i.S:37: Error: unrecognized opcode `fence.i'

I am only learning how to set up all this stuff so I may have made a small error. 

I cloned the scr1 GitHub onto my machine then copied over all of the files into my own personal GitHub. Once the files were all moved over I downloaded the GCC toolchain onto my machine and untarred it. I am using Ubuntu 22.04. Then I ran the command from the user manual: "git submodule update --init --recursive". I didn't know whether to run that command while in the original scr1 repository or while in my personal GitHub so I ran it in both. 

One thing to add is that the error also shows up when I run the make command in the original scr1 repository. 

According to the user manual, I should be able to simply run the make command while in my directory but I am met with the error above whenever I try to. Please let me know if you need any more info or if you have any solutions. Thanks for the help.

KirillSmirnov, 2023-07-31:
Please take a look at the PR: https://github.com/syntacore/scr1/pull/47
fence.i instruction has been moved to dedicated extension.

And if you use a custom gcc toolchain, you might be interested in this PR: https://github.com/syntacore/scr1/pull/49

cartersmith15, 2023-08-01:
So I noticed in the user manual that you must edit the scr1_arch_description.svh file to make a custom configuration.  Here is where I get confused. On the riscv.org site under specifications, I was reading the riscv-spec-20191213.pdf file on page 152 it says that the "G" "Standard Unprivileged Extension" has the Zifencei that I need. So then I tried editing the scr1_arch_description.svh file so that I have that extension applied but it still did not fix the error. I took a look into the root Makefile and it outlines from what I understand only certain accepted extensions. The line I am referring to reads: 

These options are for compiling tests only. Set the corresponding RTL parameters manually in the file scr1_arch_description.svh.
            # ARCH = <IMC, IC, IM, I, EMC, EM, EC, E>

So how can I get Zifencei if I can not use the extension that has Zifencei. 

Thanks for the help.

KirillSmirnov, 2023-08-01:
SCR1 implements fence.i instruction in any configuration. It is just the recent binutils (namely, assembler) do not recognize fence.i unless user explicitly tells "-march=rv32...._zifencei".

The issue is fixed (but not approved and merged yet) in my PR #47 . Please, grab the patch from that PR and see whether it fixes the issue for you.
Here is the direct link for convenience:
https://github.com/syntacore/scr1/commit/fc0664d377caa7a18cc2df6cecadc817c9f0b0fd.diff

cartersmith15, 2023-08-01:
That seems to have fixed the "unrecognized opcode `fence.i'"  error. Thank you for the help. Now I am getting an error that says the following: 

make: *** [Makefile:222: riscv_compliance] Error 2

Do you know what this means?

KirillSmirnov, 2023-08-01:
Please provide more info

cartersmith15, 2023-08-01:
Here is the whole error:

cc1: fatal error: /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/riscv_compliance/compliance_asm/I-ADD 01.S: No such file or directory
compilation terminated.
make[1]: *** [Makefile:117: /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/riscv_compliance/riscv_compliance_objs/I-ADD-01.o] Error 1
make[1]: Leaving directory '/home/csmith/scr1/sim/tests/riscv_compliance'
make: *** [Makefile:222: riscv_compliance] Error 2

KirillSmirnov, 2023-08-01:
".../I-ADD 01.S"

The filename looks suspicious. Should be a hyphen instead of space: "I-ADD-01.S"

cartersmith15, 2023-08-01:
So the makefile might have an error in it? How do I know which make file might have the spelling error?

KirillSmirnov, 2023-08-01:
These files are auto-generated from dependencies/riscv-compliance folder. Makefile does not have them hard-coded inside.

Could you please attach the entire output log from the very beginning?

cartersmith15, 2023-08-01:
After I run make this is what I get back:

rm -f /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/test_info
make: Warning: File '/home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0' has modification time 0.00037 s in the future
rm -f /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/*.hex
make -C /home/csmith/scr1/sim/tests/riscv_isa ARCH=imc
make[1]: Entering directory '/home/csmith/scr1/sim/tests/riscv_isa'
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/add.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/add.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/addi.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/addi.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/and.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/and.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/andi.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/andi.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/auipc.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/auipc.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/beq.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/beq.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bge.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bge.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bgeu.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bgeu.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/blt.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/blt.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bltu.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bltu.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bne.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/bne.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/csr.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/csr.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/fence_i.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/fence_i.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/illegal.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/illegal.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/jal.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/jal.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/jalr.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/jalr.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lb.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lb.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lbu.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lbu.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lh.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lh.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lhu.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lhu.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lui.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lui.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lw.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/lw.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/ma_addr.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/ma_addr.hex
riscv64-unknown-elf-objcopy -O verilog /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/ma_fetch.elf /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/ma_fetch.hex
[... cut ...] v_compliance_objs/I-ADD-01.o
cc1: fatal error: /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/riscv_compliance/compliance_asm/I-ADD-01.S: No such file or directory
compilation terminated.
make[1]: *** [Makefile:117: /home/csmith/scr1/build/verilator_AHB_MAX_imc_IPIC_1_TCM_1_VIRQ_1_TRACE_0/riscv_compliance/riscv_compliance_objs/I-ADD-01.o] Error 1
make[1]: Leaving directory '/home/csmith/scr1/sim/tests/riscv_compliance'
make: *** [Makefile:222: riscv_compliance] Error 2

KirillSmirnov, 2023-08-01:
:-(
attach it as a file instead of pasting output to github input form. As you can see, markdown interprepter corrupts sensitive symbols. It's very hard to analyze logs like this.

Best way to get the log is the following:
```
make clean
make 2>&1 | tee build.log
```
All data you can see on console is now written to build.log file, which you can attach easily.

cartersmith15, 2023-08-01:
I apologies.

[build.log](https://github.com/syntacore/scr1/files/12233494/build.log)

KirillSmirnov, 2023-08-01:
Thank you.

Something is wrong with the file sim/tests/riscv_compliance/Makefile

The target "cp_asm" (line 151) creates "compliance_asm" folder and then generates a bunch of asm files in it. As I can see in logs, the folder is created indeed, but no single file is created. The corresponding commands are merely ignored. Why? That's strange.

Maybe there are local changes in that Makefile? Please attach the complete output of 'git diff' (as a file).

achuykov-sc, 2023-08-02:
Hi @cartersmith15,

Could you please check that "dependencies" subfolder is not empty?
The folder "dependencies" contains multiple tests and benchmarks. And all of them (include compliance tests) are submodules (with specific version). If you do not have files there, please clone repository with submodules or initialize them. 

Best regards,
Alexander

cartersmith15, 2023-08-02:
Here is the output for git diff for @KirillSmirnov :
[git.diff.log](https://github.com/syntacore/scr1/files/12240130/git.diff.log)

I don't remember making local changes but I was troubleshooting for the past two days so it is possible I made an edit on accident.

As for @achuykov-sc question. I do have directories in the dependencies folder and files in those directories. The directorys in dependancies are : 

coremark  riscv-arch  riscv-compliance  riscv-tests

Example of files in cormark: 

barebones  core_list_join.c  core_main.c  coremark.h  coremark.md5  core_matrix.c  core_state.c  core_util.c  cygwin  docs  freebsd  LICENSE.md  linux  linux64  macos  Makefile  README.md  rtems  simple

Thanks for the help guys. :)

cartersmith15, 2023-08-02:
Also, I made a backup before editing the sim/tests/riscv_compliance/Makefile. Not sure if this is helpful but here is a diff of the backup to the original. The backup was made before adding the patch that you sent me earlier so hopefully the only changes you see are those from the patch. 
[diff.makefile.log](https://github.com/syntacore/scr1/files/12240220/diff.makefile.log)

achuykov-sc, 2023-08-07:
Hi @cartersmith15,

I did not reproduce your issue. So, here are my steps
1. Setup Linux environment with SC-DT 2012.12-SP toolchain and Verilator (in my case 4.224)
2. Clone repository
 `$ git clone --recurse-submodules git@github.com:syntacore/scr1.git`
3. Go to folder scr1 
   `$ cd scr1/`
4. Download @KirillSmirnov patch
    `$ wget -c https://patch-diff.githubusercontent.com/raw/syntacore/scr1/pull/47.patch`
5. Apply patch
   `$ git am 47.patch`
6. Make and run ISA tests
   ` make run_verilator CFG=MAX BUS=AHB TARGETS="riscv_isa" `
7. Finally, I have this

```
...

#--------------------------------------
# Summary: 56/56 tests passed
#--------------------------------------

- /home/achuykov/from_git/scr1/scr1/src/tb/scr1_top_tb_runtests.sv:199: Verilog $finish
Simulation performed on Verilator 4.224 2022-06-19 rev v4.224
                          Test               | build | simulation
                         add.hex                OK        PASS
                        addi.hex                OK        PASS
                         and.hex                OK        PASS
                        andi.hex                OK        PASS
                       auipc.hex                OK        PASS
                         beq.hex                OK        PASS
                         bge.hex                OK        PASS
                        bgeu.hex                OK        PASS
                         blt.hex                OK        PASS
                        bltu.hex                OK        PASS
                         bne.hex                OK        PASS
                         csr.hex                OK        PASS
                     fence_i.hex                OK        PASS
                     illegal.hex                OK        PASS
                         jal.hex                OK        PASS
                        jalr.hex                OK        PASS
                          lb.hex                OK        PASS
                         lbu.hex                OK        PASS
                          lh.hex                OK        PASS
                         lhu.hex                OK        PASS
                         lui.hex                OK        PASS
                          lw.hex                OK        PASS
                     ma_addr.hex                OK        PASS
                    ma_fetch.hex                OK        PASS
                        mcsr.hex                OK        PASS
                          or.hex                OK        PASS
                         ori.hex                OK        PASS
                          sb.hex                OK        PASS
                      sbreak.hex                OK        PASS
                       scall.hex                OK        PASS
                          sh.hex                OK        PASS
                       shamt.hex                OK        PASS
                      simple.hex                OK        PASS
                         sll.hex                OK        PASS
                        slli.hex                OK        PASS
                         slt.hex                OK        PASS
                        slti.hex                OK        PASS
                       sltiu.hex                OK        PASS
                        sltu.hex                OK        PASS
                         sra.hex                OK        PASS
                        srai.hex                OK        PASS
                         srl.hex                OK        PASS
                        srli.hex                OK        PASS
                         sub.hex                OK        PASS
                          sw.hex                OK        PASS
                         xor.hex                OK        PASS
                        xori.hex                OK        PASS
                         div.hex                OK        PASS
                        divu.hex                OK        PASS
                         mul.hex                OK        PASS
                        mulh.hex                OK        PASS
                      mulhsu.hex                OK        PASS
                       mulhu.hex                OK        PASS
                         rem.hex                OK        PASS
                        remu.hex                OK        PASS
                         rvc.hex                OK        PASS
```

Best regards,
Alexander

closed 2023-08-23 by achuykov-sc