syntacore/scr1 #51: Error running Coremark benchmark

nerdylye, 2023-08-29:
Hi, I'm trying to run coremark benchmark but this error starts showing up. I'm using Verilator v4.102. The riscv_isa runs fine. May I know if there is something I need to do to run coremark? Thanks in advance :D 

[image]
[image]

achuykov-sc, 2023-08-29:
Hi @nerdylye,

Could you please clarify, how did you build it (what is your command line)? Which is your toolchain? 

Best regards,
Alexander

nerdylye, 2023-08-29:
Hi @achuykov-sc, 

The command line is `make TARGETS='coremark'` and the toolchain is build from the official RISC-V GCC toolchain repo.

achuykov-sc, 2023-08-29:
Hi @nerdylye,

We are recommending to use Syntacore toolchain from: https://syntacore.com/page/products/sw-tools

Here are steps to run Coremark
1. Setup Linux environment with SC-DT toolchain and Verilator 
    * `SC_DT_PATH` -- SC-DT root folder 
    * `SC_BSP_PATH` -- Path to BSP location (by default, `${SC_DT_PATH}/workspace/bsp`)
    * `SC_CLANG_PATH` -- Path to LLVM/clang toolchain (by default, `${SC_DT_PATH}/llvm`)
    * `SC_CMAKE_TOOLCHAINS` -- Path to CMake toolchain files (by default, `${SC_DT_PATH}/workspace/cmake-toolchains`)
    * `SC_GCC_PATH` -- Path to GCC toolchain (by default, `${SC_DT_PATH}/riscv-gcc`)
    Update `PATH` environment variable, append or prepend it with `${SC_GCC_PATH}/bin`, `${SC_CLANG_PATH}/bin` and `${SC_DT_PATH}/tools/bin`. If you would use Eclipse-base IDE, add these to your `PATH`: `${SC_DT_PATH}` and `${SC_DT_PATH}/eclipse`. 

2. Clone repository
   `$ git clone --recurse-submodules git@github.com:syntacore/scr1.git`

3. Go to folder `scr1`
    `$ cd scr1/`

4. Download patch
    `$ wget -c https://patch-diff.githubusercontent.com/raw/syntacore/scr1/pull/47.patch`

5. Apply patch
    `$ git am 47.patch`

5. Make and run Coremark 
    `$ make run_verilator CFG=MAX BUS=AHB TARGETS="coremark"`

Here is my output
```
... 
---Test:                     coremark.hex
CoreMark 1.0
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 4870
Total time (secs): 0
ERROR! Must execute for at least 10 secs for a valid result!
Iterations       : 1
Compiler version : GCC12.2.1 20221101
Compiler flags   : -O2 -funroll-loops -fpeel-loops -fgcse-sm -fgcse-las
Memory location  : STATIC
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0xe714
Errors detected
Test passed

#--------------------------------------
# Summary: 1/1 tests passed
#--------------------------------------

- /home/achuykov/from_git/scr1/scr1/src/tb/scr1_top_tb_runtests.sv:199: Verilog $finish
Simulation performed on Verilator 5.014 2023-08-06 rev v5.014
                          Test               | build | simulation
                    coremark.hex                OK        PASS
```

Best regards,
Alexander

AntonTimofeev, 2023-08-29:
Patch fixes the issue.
Tested with `202308-sc-dt`

`make run_verilator CFG=MAX BUS=AHB TARGETS="coremark" SIM_BUILD_OPTS=--timing`
```
---Test:                     coremark.hex
CoreMark 1.0
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 4870
Total time (secs): 0
ERROR! Must execute for at least 10 secs for a valid result!
Iterations       : 1
Compiler version : GCC12.2.1 20221101
Compiler flags   : -O2 -funroll-loops -fpeel-loops -fgcse-sm -fgcse-las 
Memory location  : STATIC
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0xe714
Errors detected
Test passed

#--------------------------------------
# Summary: 1/1 tests passed
#--------------------------------------

- /home/***/scr1/src/tb/scr1_top_tb_runtests.sv:199: Verilog $finish
Simulation performed on Verilator 5.011 devel rev v5.010-114-g4b7b185d0 
                          Test               | build | simulation 
                    coremark.hex		OK	  PASS
```

achuykov-sc, 2023-08-29:
Hi @AntonTimofeev,

Thanks for confirmation. I'm closing the issue as fixed. 

BR,
Alexander

nerdylye, 2023-08-30:
Hi @achuykov-sc , just curious where can I get the coremark scores?

closed 2023-09-25 by achuykov-sc