syntacore/scr1 #52: Coremark testcase still failed after added #9 issue solution

weipo790, 2023-09-30:
Hello,
I am running the benchmark of coremark and found some errors after following the #9 and get the results below:
First, I try the command: 
$make run_vcs CFG=CUSTOM BUS=AXI ARCH=I VECT_IRQ=0 IPIC=0 TCM=0 TARGETS="coremark" ITERATIONS=31000 
and the iteration times doesn't changed (still Iterations=1 and shows the results as below).
[image]
So I directly change the iteration time in /coremark/makefile like below.
[image]
This time it does change, but was not the result I expected. How can I fix it to get the pass result?
[image]

Thank you!

achuykov-sc, 2023-10-02:
Hi @weipo790,

The timeout is a parameter in testbench:
* for AXI: https://github.com/syntacore/scr1/blob/master/src/tb/scr1_top_tb_axi.sv#L21
* for AHB: https://github.com/syntacore/scr1/blob/master/src/tb/scr1_top_tb_ahb.sv#L22

Please adjust it according to your requirements. And one extra note: even if you run enough iterations, you will not see prints with score. Because of: 
* the `HAS_FLOAT` is not defined, and it is required for print
* the CoreMark's Makefile does **not** enable define `PERFORMACE_RUN=1`

My suggestions: 
* modify CoreMark's Makefile and portable code in SCR1 
--OR-- 
* use CoreMark from SC-DT sample application (but you should use custom BSP for it) 

Best regards,
Alexander

weipo790, 2023-10-02:
Hi @achuykov-sc ,
  I appreciate your quick comment, but I still have one question that needs to be clarified.
1. you say that the `HAS_FLOAT` is not defined, and it is required for print. 
Can I interpret it this way? Because the SCR1 core only supports RISC-V32 integer computation, and at the same time, If I need Coremark score results that require defining `HAS_FLOAT,` and `HAS_FLOAT` requires floating-point operations to be supported. However, SCR1 does not support floating-point operations, so it cannot use the complete Coremark benchmark to calculate Coremark/sec(which means we can't `#define HAS_FLOAT 1` so there is no score result). However, we can execute other instructions related to integer operations with define HAS_PRINT 0. Therefore, in the SCR1 test, Coremark only runs the integer computation part of the benchmark without giving the score.

BTW in [core_main.c](https://github.com/eembc/coremark/blob/41537ea30b0104438b4ff993e7d349af26900acf/core_main.c) the `time_in_secs(total_time)` is always equal to zero no matter how many iterations I set to the coremark makefile, which cause the error result on my scr1 coremark simulation. I'm just puzzled why I can't obtain results in my simulation similar to those shown in the [presentation](https://www.youtube.com/watch?v=dLiKDcXxK3U&t=68s&ab_channel=FOSSiFoundation).

[image]

achuykov-sc, 2023-10-03:
Hi @weipo790,

The `HAS_FLOAT` impacts only on printing. So, you could comment out the snippet on your last screenshot. Then, you should modify CoreMark's Makefile to set more iterations and `-DPERFORMACE_RUN=1`. 

Because SCR1 is simple core with in-order execution and there are no caches there, the number of iterations may not impact on execution time. You may decrease number of iterations (if you would, of course) and have stable result. 

So, than you could compute CoreMark score as `[number of iterations] * [scale factor] / [Total ticks]`. The `[scale factor]` is 10000 because time is calculated by timer: each tick is equal 100 CPU cycles. 

I’m looking how to achieve best score and update you later. 

Best regards,
Alexander

closed 2023-11-28 by achuykov-sc

arunvthampi, 2024-03-29:
Hi @achuykov-sc, 
You said to modify CoreMark's makefile to PERFORMACE_RUN=1. Would it be possible to tell me how to change that parameter. After making changes in Makefile and https://github.com/syntacore/scr1/blob/master/src/tb/scr1_top_tb_axi.sv#L21, https://github.com/syntacore/scr1/blob/master/src/tb/scr1_top_tb_ahb.sv#L22, its still showing the same error