syntacore/scr1 #53: Invalid result of `DIV` operation

chykon, 2023-10-04:
Hello!

While testing the [iALU](https://github.com/syntacore/scr1/blob/master/src/core/pipeline/scr1_pipe_ialu.sv) module, I encountered an incorrect result from the `DIV` operation. The expected result of signed integer division (operand 1 is `0xAAAA_AAAA`, operand 2 is `0x5555_5555`) is `2`, but iALU produces `0xFFFF_FFFF`.

Waveforms:

[image]

Example to reproduce (cocotb):

```python
from typing import Any

import cocotb
from cocotb.clock import Clock
from cocotb.runner import get_runner
from cocotb.triggers import FallingEdge

@cocotb.test()
async def example_test(dut: Any):
    cocotb.start_soon(Clock(dut.clk, 2).start(start_high=False))

    await FallingEdge(dut.clk)
    await FallingEdge(dut.clk)

    # Reset

    dut.rst_n.value = 0
    dut.exu2ialu_main_op1_i.value = 0
    dut.exu2ialu_main_op2_i.value = 0
    dut.exu2ialu_cmd_i.value = 0  # SCR1_IALU_CMD_NONE
    dut.exu2ialu_rvm_cmd_vd_i.value = 0

    await FallingEdge(dut.clk)
    await FallingEdge(dut.clk)

    dut.rst_n.value = 1

    # Test

    OP1 = 0b10101010101010101010101010101010  # 0xAAAA_AAAA
    OP2 = 0b01010101010101010101010101010101  # 0x5555_5555

    dut.exu2ialu_main_op1_i.value = OP1
    dut.exu2ialu_main_op2_i.value = OP2
    dut.exu2ialu_cmd_i.value = 19  # SCR1_IALU_CMD_DIV
    dut.exu2ialu_rvm_cmd_vd_i.value = 1

    await FallingEdge(dut.clk)

    while dut.ialu2exu_rvm_res_rdy_o.value != 1:
        await FallingEdge(dut.clk)

    assert dut.ialu2exu_main_res_o == (
        OP1 // OP2
    )  # actual: `0xFFFF_FFFF`, expected: `2`

def run():
    runner = get_runner("verilator")

    runner.build(
        verilog_sources=["src/core/pipeline/scr1_pipe_ialu.sv"],
        includes=["src/includes"],
        hdl_toplevel="scr1_pipe_ialu",
        build_args=["-Wno-WIDTHEXPAND", "-Wno-CASEINCOMPLETE", "--trace"],
    )

    runner.test(hdl_toplevel="scr1_pipe_ialu", test_module="example")

if __name__ == "__main__":
    run()
```

---

* OS: Fedora 38
* Verilator 5.016 2023-09-16 rev v5.014-149-g57c816f90
* Python 3.11.4
* cocotb 1.8.0

achuykov-sc, 2023-10-04:
Hi @chykon,

Please apologize, but I do not agree with you: the mistake in your python testebench. The Python does not care about the length of the integer variable, so `OP1` is 64-bit integer. And when you do `OP1 // OP1` you have division of two **positive** numbers and result is 2. 

If you run the code with 32-bit variables, you have correct result `-1` because `OP1 = -1431655766` and `OP2 = 1431655765`. You could test this even on x86. 

```C
#include <stdint.h>
#include <stdio.h>

void div_test(void)
{
    volatile int32_t OP1 = (int32_t)0xAAAAAAAA;
    volatile int32_t OP2 = (int32_t)0x55555555;
    volatile int32_t res = OP1 / OP2;
    printf("%d / %d = %d\n", OP1, OP2, res);
}

int main()
{
    div_test();
    return 0;
}
```

Result: 
```
-1431655766 / 1431655765 = -1
```

The online: [online compler](https://godbolt.org/#g:!((g:!((g:!((h:codeEditor,i:(filename:'1',fontScale:14,fontUsePx:'0',j:1,lang:___c,selection:(endColumn:2,endLineNumber:16,positionColumn:1,positionLineNumber:1,selectionStartColumn:2,selectionStartLineNumber:16,startColumn:1,startLineNumber:1),source:'%23include+%3Cstdint.h%3E%0A%23include+%3Cstdio.h%3E%0A%0Avoid+div_test(void)%0A%7B%0A++++volatile+int32_t+OP1+%3D+(int32_t)0xAAAAAAAA%3B%0A++++volatile+int32_t+OP2+%3D+(int32_t)0x55555555%3B%0A++++volatile+int32_t+res+%3D+OP1+/+OP2%3B%0A++++printf(%22%25d+/+%25d+%3D+%25d%5Cn%22,+OP1,+OP2,+res)%3B%0A%7D%0A%0Aint+main()%0A%7B%0A++++div_test()%3B%0A++++return+0%3B%0A%7D'),l:'5',n:'0',o:'C+source+%231',t:'0')),k:32.331271688264216,l:'4',n:'0',o:'',s:0,t:'0'),(g:!((h:compiler,i:(compiler:cgsnapshot,deviceViewOpen:'1',filters:(b:'0',binary:'1',binaryObject:'1',commentOnly:'0',debugCalls:'1',demangle:'0',directives:'0',execute:'0',intel:'0',libraryCode:'1',trim:'1'),flagsViewOpen:'1',fontScale:13,fontUsePx:'0',j:1,lang:___c,libs:!(),options:'-O2',overrides:!(),selection:(endColumn:1,endLineNumber:1,positionColumn:1,positionLineNumber:1,selectionStartColumn:1,selectionStartLineNumber:1,startColumn:1,startLineNumber:1),source:1),l:'5',n:'0',o:'+x86-64+gcc+(trunk)+(Editor+%231)',t:'0')),k:34.33539497840246,l:'4',n:'0',o:'',s:0,t:'0'),(g:!((h:output,i:(compilerName:'x86-64+gcc+(trunk)',editorid:1,fontScale:14,fontUsePx:'0',j:1,wrap:'1'),l:'5',n:'0',o:'Output+of+x86-64+gcc+(trunk)+(Compiler+%231)',t:'0')),k:33.33333333333333,l:'4',n:'0',o:'',s:0,t:'0')),l:'2',n:'0',o:'',t:'0')),version:4)

Best regards,
Alexander

chykon, 2023-10-04:
Yes, that's right. Thanks for pointing out my mistake.

I think the issue can be closed.