syntacore/scr1 #55: How does it work for sc_print to print words in terminal when running HDL simulator?

weipo790, 2024-01-24:
Hi,

I have been exploring how to compile sc_print.c into a hex file using the RISCV toolchain and then load it into the memory model using the system call $readmemh. However, I am curious about how the print text is displayed on the terminal when using an HDL simulator (e.g., vcs).
[image]
I have looked through scr1_top_tb_runtests.sv and other related files but haven't been able to find a $display function or any reference to how SC_SIM_OUTPORT (0xf0000000) is handled. Could you please provide some insights or guidance on how this process works?

Thank you very much for your time and assistance. I greatly appreciate any help you can offer.

Best regards,
WeiPo

achuykov-sc, 2024-01-24:
Hi @weipo790,

The print character function is implemented in file `sim/tests/common/sc_print.c`, and it writes characters by address `SC_SIM_OUTPORT` = 0xf0000000 (defined is same file). This address is `SCR1_SIM_PRINT_ADDR`, defined in `src/includes/scr1_arch_description.svh`. Then you could search for `SCR1_SIM_PRINT_ADDR` in files `src/tb/scr1_memory_tb_axi.sv` and `src/tb/scr1_memory_tb_ahb.sv` to find entries. For example, for AHB: https://github.com/syntacore/scr1/blob/7432bd0a61ecc3181c441ed6c3dd37d394b21c2a/src/tb/scr1_memory_tb_ahb.sv#L564

Here is the snippet: 
```Verilog
            case (dmem_ahb_addr)
                // Printing character in the simulation console
                SCR1_SIM_PRINT_ADDR : begin
                    $write("%c", dmem_hwdata[7:0]);
                end
```

I hope this helps you. 

Best regards,
Alexander

weipo790, 2024-01-24:
Hi @achuykov-sc ,
  I got it. Thanks a lot. 

B.R.
Wei-Po

closed 2024-01-24 by weipo790