syntacore/scr1 #59: Error when running Dhrystone

lokki11199, 2024-02-24:
Hello,
I get the following error when running Dhrystone:
[image]
I am using your Pre-built RISC-V GCC toolchain from http://syntacore.com/page/products/sw-tools.
At the same time, all other tests run without errors

Thank you!

achuykov-sc, 2024-02-26:
Hi @lokki11199,

Let me apologize for inconvenience. It is known issue. To fix it, please open `sim/tests/common/common.mk` and add `--specs=nano.specs` to the line 15:
```
LDFLAGS   ?= -nostartfiles -nostdlib -lc -lgcc -march=rv32$(ARCH)_zicsr_zifencei -mabi=$(ABI) --specs=nano.specs
```

We will fix it permanently in the next SCR1 release. 

Best regards,
Alexander

lokki11199, 2024-02-26:
Thanks for the reply! It works.

closed 2024-02-26 by lokki11199