syntacore/scr1 #66: store instructions may fail when a CSR is written

gambaman, 2024-04-03:
Hello. It looks store instructions may fail when a CSR is written. For example, try this code in commit 7432bd0a61ecc3181c441ed6c3dd37d394b21c2a:

`#include "sc_print.h"
#include "csr.h"

#define mstatus 0x300

int test_instruction(char* name,uint32_t result,uint32_t expected)
{
  sc_printf("testing instruccion %s. result:%#08x expected:%#08x.",name, result, expected);
  if(result==expected)
    {sc_printf("\n");return 0;}
  //else
  sc_printf("    FAILS\n");
  return 1;
}

void simulate_little_endian_store(unsigned int size, uint32_t value, void* address, int32_t offset)
{
  for(int i=offset;i<size+offset;i++)
  {
    *( ((uint8_t*)address) +i) = (uint8_t)value;
    value>>=8;
  }
}

#define simulate_little_endian_SW(value,address,offset) simulate_little_endian_store(4,value,address,offset)
#define simulate_little_endian_SH(value,address,offset) simulate_little_endian_store(2,value,address,offset)
#define simulate_little_endian_SB(value,address,offset) simulate_little_endian_store(1,value,address,offset)

void little_endian_SH(uint32_t value, void* address, int32_t offset)
{
    volatile uint16_t* efective_address= (uint16_t*) (address+offset);
    write_csr(mstatus,read_csr(mstatus));
    *efective_address=value;
    sc_printf("Extrange: It does not fail is this sc_printf call is removed or if no CSR is written %d%d%d\n", 0, 0, 0);
}

int main()
{
  int fails=0;
  uint32_t result,expected;

  result=0x88888888;expected=0x88888888;
  little_endian_SH(0x76543210,&result,0);
  simulate_little_endian_SH(0x76543210,&expected,0);
  fails|=test_instruction("SH",result,expected);

  result=0x88888888;expected=0x88888888;
  little_endian_SH(0x76543210,&result,2);
  simulate_little_endian_SH(0x76543210,&expected,2);
  fails|=test_instruction("SH",result,expected);

  return fails;
}
`

achuykov-sc, 2024-04-04:
Hi @gambaman,

It seems a compiler issue, the application becomes overoptimized. I'm looking into details to find the root cause and stable fix. 

BR,
Alexander

kv-sc, 2024-04-04:
You have strict aliasing violation between:

```
uint32_t result
volatile uint16_t* efective_address
```

You can compile with -fno-strict-aliasing all will work.

Reduced x86 example: https://godbolt.org/z/Eqao1K9T1

gambaman, 2024-04-04:
> You have strict aliasing violation between:
> 
> ```
> uint32_t result
> volatile uint16_t* efective_address
> ```
> 
> You can compile with -fno-strict-aliasing all will work.
> 
> Reduced x86 example: https://godbolt.org/z/Eqao1K9T1

Hello, Kv-sc. The example code I used is helpless. I could just remove the CSR access or the sc_printf call and it will work. That is not the problem. The problem is that (apparently) scr1 is not working properly.

gambaman, 2024-04-07:
> You have strict aliasing violation between:
> 
> ```
> uint32_t result
> volatile uint16_t* efective_address
> ```
> 
> You can compile with -fno-strict-aliasing all will work.
> 
> Reduced x86 example: https://godbolt.org/z/Eqao1K9T1

Hello, Kv-sc. You are right. Because of the strict aliasing violation the compiler use the constant 0x88888888 as actual parameter in when calling test_instruction:

`
  481144:	21060613          	add	a2,a2,528 # 88883210 <__C_STACK_TOP__+0x883f3210>
  481148:	88858593          	add	a1,a1,-1912 # 88888888 <__C_STACK_TOP__+0x883f8888>
  48114c:	b0418513          	add	a0,gp,-1276 # 482124 <__clz_tab+0x1ac>
  481150:	35d9                	jal	481016 <test_instruction>`

Thank you very much

gambaman, 2024-04-07:
> Hi @gambaman,
> 
> It seems a compiler issue, the application becomes overoptimized. I'm looking into details to find the root cause and stable fix.
> 
> BR, Alexander

As noted by you and by Kv-sc, it is a compiler issue.

closed 2024-04-07 by gambaman