syntacore/scr1 #67: mstatus register resets once the breakpoint hits

jyothishdevan, 2024-04-15:
While debugging the code, the mstatus register resets once the breakpoint hits. 
In my case, I have enabled global interrupt hence the mie bit is 1 in the mstatus register. But after the breakpoint hits the mie bit becomes zero and the global interrupts disabled. So I didn't get any further interrupts after breakpoint hits.

achuykov-sc, 2024-04-15:
Hi @jyothishdevan,

Could you please add more details? 

1. What kind of BP do you set: hardware or software? 
2. And how do handle the breakpoint? (in application or via debugger) 
3. Do you run on board or RTL-sim? 

If you could have testcase to share, please let share it. 

Best regards,
Alexander

achuykov-sc, 2024-04-15:
Hi @jyothishdevan,

I did quick checks but with another adapter and GDG from SC-DT 2024.02. And there are no issues:
1. During interrupt handling the `mstatus=0x1880` (MIE = 0) as expected 
2. But after complete interrupt handler exit (in `main` function) the `mstatus=0x1888` (MIE=1)

Could you please check that you are not disabling interrupt in handler? And please double check that your handler's last instruction is `mret`. 

Best regards,
Alexander

jyothishdevan, 2024-04-15:
Hi @achuykov-sc 
It is working as you have explained for the case of interrupts. But in my case, the mie bit becomes zero when the break point hits. In interrupt handler I am not disabling interrupt and mret is the last instruction in interrupt handler. I am keeps getting interrupts and runs interrupt handler if there is no breakpoint(or no breakpoint hits)

Thanks and regards,
Jyothish

achuykov-sc, 2024-04-15:
Hi @jyothishdevan,

Thanks for quick update. What is your SCR1 configuration? 
And could you please clarify SEGGER Embedded Studio version? 

I will try to run sample code on JLink within next days. 

Best regards,
Alexander

jyothishdevan, 2024-04-15:
Hi @achuykov-sc 
SCR1 configuration is RV32IC base
Segger embedded studio version 7.22

Thanks for the support
Regards,
Jyothish

achuykov-sc, 2024-04-16:
Hi @jyothishdevan, 

I had tests with J-Link Ultra connected to FPGA, and it work fine. But I did tests with SC-DT GDB. If you could share Embedded Studio reproducer, I will double check. 

Best regards,
Alexander

jyothishdevan, 2024-04-17:
Hi @achuykov-sc 
I didn't understand what is meant by embedded studio reproducer?

achuykov-sc, 2024-04-17:
Hi @jyothishdevan,

Please apologize for unclarity. I mean the SEGGER Embedded Studio project which reproduces the issue. Are you able to share it? 
Because I did not reproduce your issue on my setup.

BTW, please make sure that you correctly configure `MTVEC` register on boot (if you do this). In the BASE configuration the number of configurable bits in `MTVEC` is **16** by default. It means you can configure only high 16-bits of the register, but low 16-bit are fixed in hardware design. 

Best regards,
Alexander

closed 2024-05-28 by achuykov-sc