syntacore/scr1 #75: Issue with reset domain crossing in TAPC

brunopen, 2025-04-22:
In SCR1 external architecture specification, Table 78, pwrup_rst_n input Power-up asynchronous reset. 
This async input goes through i_tapc_rstn_and2_cell in scr1_top_ahb.sv, gets out as tapc_trst_n and from there goes in scr1_tapc where it is used as reset forall the always_ff in it. 
It means we are getting an async input reaching without synchronization FF. 

Is this expected? 
How is this supposed to be a clean reset-domain-crossing ? 
Thanks, Bruno

achuykov-sc, 2025-04-29:
Hi @brunopen,

Let me look in details your question. I will keep you updated. 

Best regards,
Alexander

achuykov-sc, 2025-05-12:
Hi @brunopen,

Thank you for patience. Here is the answer from my colleague:

Yes, this is an expected connection. Power-Up-Reset is an unconditional reset for all logic within SCR1, thus it should influence every FF inside including TAPC. So, there are no doubts regarding ASSERTION of the Power-Up-Reset and its domain of applicability so far, isn't it? Hopefully, I was enough clear to express: Power-Up-Reset and reset-domain-crossing -- joint consideration of those two have no sense, as Power-Up-Reset domain is all SCR1 and there is nothing to cross.

There are no issues with asynchronous entering into reset state by Power-Up-Reset cause. But some issues could arise during NEGATION of the Power-Up-Reset and exiting of logic from reset state to some working one. And here we have the only relation to the synchronization topic: internal Power-Up-Reset negation must be synchronized with clock somehow. For CPU logic and TAPC this requirement is met in different ways:

Core/CPU: by a reset sync cell - e.g., for AHB_TOP this is i_pwrup_rstn_reset_sync;
JTAG/TAPC: by recommendation or even requirement to hold TMS=1 around Power-Up-Reset's transition from 0 to 1 (this will effectively hold TAPC FSM in Test-Logic-Reset state); or, additionally, by a requirement to have a reasonable delay between Powering-Up and a start of JTAG operations. The 1st requirement is also aligned with common JTAG HW requirement to provide Pull-Up resistors for TMS, TDI, TDO, TRSTn, and Pull-Down resistor for TCK.

Best regards,
Alexander

closed 2025-07-28 by achuykov-sc